jantenhove/GoodWeLogger #6: Not an issue

slampt, 2017-07-19:
Hello, not an issue but I wanted to get in touch and ask you a few questions about this logger via email.

Thanks!

jantenhove, 2017-07-19:
You can e-mail me on: goodwelogger@forward.cat

closed 2017-07-19 by jantenhove

wurtel2, 2017-10-16:
Hi, I tried mailing you on the above address, but I got the following unhelpful error:

 Failed recipient(s):                                                                                
 goodwelogger@forward.cat                                                                            
                            
 Error message:                                                        
 null                                                                

I was wondering if you had any idea how to connect if the inverter (GW10KN-DT) shows 2 pairs of RS485 connections, a RX and a TX pair. Are 2 RS485 interfaces needed?

jantenhove, 2017-10-16:
Only one RS485 connection is needed. The second RS485 connection is for connecting to the next inverter so they can be daisy-chained.
The forward e-mail address you've mailed is a temporary email address. Commenting and questions can be posted here.

wurtel2, 2017-10-16:
There are 2 physical connectors, with each connector showing 2 pairs of RS485 pins. This is a grab of the installation manual where it's described: http://corky.wurtel.net/~paul/rs485-rj45-2.gif
I'll have a look in person at the unit this weekend to see what's actually there :-)

talondnb, 2018-07-14:
Hi!

I’m interested in using this for a SolarRiver inverter. Would the implementation be the same?