jantenhove/GoodWeLogger #15: Powering the ESP8266

danjenkins, 2018-08-25:
Hi @jantenhove ,

Firstly - great project - works brilliantly. Now have all my solar values going into home assistant which then get logged to grafana - so can do some nice visuals & do automations based on generation etc.

I have a question though - how do you power your esp8266? I tried connecting a usb cable to my GW3600D-NS's USB port and into the d1 mini but no juice appears from the USB port. How best to power it? Or should I be getting power through the uart?

Cheers!

Dan

meteosat007, 2018-08-28:
Hi Dan,

I have a GoodWe 1000NS inverter also known as a Replus 1000TL2E with Wifi, USB and RS485 and I would be keen to hear your experiences of getting the data into Emoncms as its the last thing I would like to do on my system that I have had up and running for 4 years now. Any advice welcomed as I have tried a few options from Github with no success on USB or RS485.

Alan

jantenhove, 2018-10-19:
I tried powering the ESP from the USB port, but it does not provide power. I now use a USB charger.

closed 2018-10-19 by jantenhove

danjenkins, 2018-10-19:
Yup, I ended up running a longer cable to the inverter and run the logger
off usb in my server rack

ThinkPadNL, 2020-04-22:
Hi,

Have a look at issue #25 that i just opened. I think powering the ESP8266 from the inverter should be possible, but i am not 100% sure about it yet.