jantenhove/GoodWeLogger #17: Getting E-Total

richardmh, 2018-10-18:
Hi

This works great on my GW10KN-DT to MQTT but I would also like to get the E-Total but so far not successful.   Main trouble is your code is far too clever for me and I don't really understand it, but this is what I did so far:

set isDTseries to true so am getting all values correctly
added some new vars and mqtt topic lines, and I am getting new data...

in goodWeCommunicator.cpp starting about line 346

	//TODO: Get the other values too
	// inverter->temp = bytesToFloat(data + dtPtr, 10);   dtPtr += inverter->isDTSeries ? 34 : 26; //orig
	//new other values below
	inverter->temp = bytesToFloat(data + dtPtr, 10); dtPtr += 2;
	dtPtr += 4;  //skip over error msg H & L
	inverter->etoth = bytesToFloat(data + dtPtr, 10);    dtPtr += 2;
	inverter->etotl = bytesToFloat(data + dtPtr, 10);    dtPtr += 2;
	inverter->htoth = bytesToFloat(data + dtPtr, 10);    dtPtr += 2;
	inverter->htotl = bytesToFloat(data + dtPtr, 10);    dtPtr += 2;
	dtPtr += inverter->isDTSeries ? 20 : 12 ; //skip over next fault messages
	//new other values above
	inverter->eDay = bytesToFloat(data + dtPtr, 10);

I realise that h & l must be combined somehow into the vars eTotal and hTotal you already provided but I thought I would start by just replicating your code to see what I got but the resulting data makes no sense:
etoth = 0.0
etotl = 504.5
Inverter screen shows 7401.6 kWh

Help so I can get the proper values would be most welcome!

Regards

Richard

jantenhove, 2018-10-19:
Instead of using `bytesToFloat` on `etoth` and `etotl`, you need to shift the high bytes in front off the low bytes.
Something like (typed without checking syntax):
`float eTotal =  float(((unsigned int)etoth[0] << 24) | ((unsigned int)etoth[1] << 16) | ((unsigned int)etotl[1] << 8) 
 | etotl[1])`

richardmh, 2018-10-19:
Excellent, THANKYOU, it works.  (Still don't really understand bit shifting and xORing though).

For the record, this is what I did:

	//TODO: Get the other values too
	// inverter->temp = bytesToFloat(data + dtPtr, 10); dtPtr += inverter->isDTSeries ? 34 : 26; //orig
	//new other values below
	inverter->temp = bytesToFloat(data + dtPtr, 10); dtPtr += 2;
	dtPtr += 4;  //skip over error msg H & L
	inverter->eTotal = fourBytesToFloat(data + dtPtr, 10);    dtPtr += 4;
	inverter->hTotal = fourBytesToFloat(data + dtPtr, 1);    dtPtr += 4;
	dtPtr += inverter->isDTSeries ? 20 : 12 ; //skip over next fault messages
	//new other values above
	inverter->eDay = bytesToFloat(data + dtPtr, 10);		

and the new function fourBytesToFloat looks like:

	float GoodWeCommunicator::fourBytesToFloat(char * bt, char factor)
	{
	//convert 4 byte to float by converting to int and then dividing it by factor
	return float( ((unsigned int)bt[0] << 24) | ((unsigned int)bt[1] << 16) | ((unsigned int)bt[2] << 8) | bt[3]) / factor;
	}

Obviously there are one or two further things to do - set up MQTTPublisher.cpp for eTotal and hTotal but the others will show up at compile.

Richard

closed 2018-10-19 by richardmh