jantenhove/GoodWeLogger #18: Daylight saving time error in PVoutput posts

bram2202, 2018-10-31:
After the change in DST the posts to PVoutput keeps posting the old datetime.

For example: last weekend DST ended, correct datetime for me is UTC+1.
But PVoutput receives UTC+2, and thereby show the data in the live chart 1 hour in the future.
Screenshot taken at **11:30**

[image]

Workaround probably is to reset the board, but you need to do this before/after power is generated, otherwise PVoutput wil receive multiple data for the same datetime

jantenhove, 2018-10-31:
Time zone settings are hardcoded for now: https://github.com/jantenhove/GoodWeLogger/blob/4a638a97256bc0c7d88c07dab83a9c1134987bc2/Settings.example.h#L38
PVoutput only accepts local time instead of UTC (or any other standard).

bram2202, 2018-10-31:
Ok, so i need to change TIMEZONE every 6 months?

jantenhove, 2018-10-31:
Yes, or ask pvoutput to support UTC uploads. They already know the region you're in so they can convert the time more easily.

jantenhove, 2018-10-31:
When you donate to pvoutput there is a 'Adjust Time' setting for each of your systems to correct for systems that don't support DST. See: https://pvoutput.org/help.html#live-settings-time-adj

closed 2019-08-12 by jantenhove