jantenhove/GoodWeLogger #19: Inverter Data to Serial Port

EHUbels, 2019-01-27:
Build the GoodweLogger today and it basically run direct as expected. I'm not interested into posting the data on PVOutput, so I left all those items blank and data is visible as it should, in the serial monitor, with the necessary HEX codes of course. My question is how can I make the data readable in the serial port. Obviously I looked into the code, tried some things but mostly lead to not being able to get the data anymore. Is there a way to take the data from the handlingincoming data from the goodewelogger.cpp?

Looking forward to hear from you.

sircuri, 2019-01-27:
There is a function "getInvertersInfo" on the GoodWeCommuncation.cpp file. It contains all the discovered inverters with the data in it. Simple call that function after you called the handleData function.

EHUbels, 2019-01-27:
Dear Jan,

Thanks, I’ll have a look at it, need to wait to get the converter online tomorrow.

Thanks,

Erik

EHUbels, 2019-01-28:
Arjen or Jan,

Thanks you so much for responding, really appreciate it. I have tried many different things today but I’m going trough a steep learning curve in C++.

Looking at the GoodWeCommunication.pp I understand that ones the converter is registered it starts handling the data by the handleIncomingInformation function, than the struct,  based on the actual inverters address starts to fill from the individual items (inverter->vpv1 = bytesToFloat(data, 10);, that is after it converted the two HEX bytes to the actual float.

I’ve tried many different ways to read the individual items from the main loop, but did not succeed. I need the individual items to build a JSON formatted message.
Any help to read the individual items (one example obviously would do it) into the main loop would be highly appreciated.

Regards,

Erik

jantenhove, 2019-07-18:
You can create a file similar to MQTTPublisher that, instead of sending it with MQTT, prints it on the serial port. 
When you created something, please send a PR, so I can add it.

closed 2019-07-18 by jantenhove