jantenhove/GoodWeLogger #21: MQTT data stays on latest value when inverter is offline

VelocityMaximus, 2019-07-09:
The MQTT data like vpv and vac stay on the latest value when the inverter goes offline.

It would be nice to show the real data and after a time-out the data will be zero.

Also the daily generated value should be reset at night for convenience.
Now it shows the data from the day before and resets when the inverter boots up.

jantenhove, 2019-07-17:
I will look into this!
When the inverter goes offline, reset all variables to zero, except for eDay. The eDay will then be resetted at midnight.

VelocityMaximus, 2019-08-11:
Latest version has a new bug: MQTT message: goodwe/96000DTUXXXXXX/online stays 1 even if the inverter went offline.

jantenhove, 2019-08-12:
I can't see any recent change that results in that behavior. I will make a change that sends the offline message when the inverter isn't registered completely. Hopefully that will fix it.

commit d85c631a (2019-08-14): Auto reset inverter values + auto reset eday at 0:00. Fixes #21
closed 2019-08-14 by jantenhove