jantenhove/GoodWeLogger #23: Error in SoftwareSerial during compiling

UnrealKazu, 2019-10-31:
When compiling the latest master I get the following error for the `SoftwareSerial` dependency:

```
In file included from sketch\GoodWeCommunicator.h:3:0,

                 from sketch\GoodWeCommunicator.cpp:1:

C:\Users\X\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/SoftwareSerial.h:113:2: error: 'ssize_t' does not name a type

  ssize_t m_swsInstsIdx = -1;

  ^
```

I am using the latest version of the Arduino IDE (1.8.10), as well as all dependent libraries mentioned in the README. 

As you can see, the version of the `SoftwareSerial` library that is used is part of 2.5.2 of the esp8266 package. Compiling an older commit does work, for example, [this](https://github.com/bram2202/GoodWeLogger) fork.

chemuki205, 2020-03-19:
Hi

I have a similar problem compiling the code and I don't know how to solve it, I would appreciate any help to be able to use it in two goodwe inverters.

```
Arduino:1.8.12 (Windows 10), Tarjeta:"LOLIN(WEMOS) D1 mini Lite, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 1MB (FS:64KB OTA:~470KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

sketch\GoodWeCommunicator.cpp: In member function 'void GoodWeCommunicator::start()':

GoodWeCommunicator.cpp:14:91: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(int&, int&, bool, const int&)'

  goodweSerial = new SoftwareSerial(settings->RS485Rx, settings->RS485Tx, false, BufferSize); // (RX, TX. inverted, buffer)

                                                                                           ^

sketch\GoodWeCommunicator.cpp:14:91: note: candidates are:

In file included from sketch\GoodWeCommunicator.h:3:0,

                 from sketch\GoodWeCommunicator.cpp:1:

C:\Users\Chema\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\SoftwareSerial\src/SoftwareSerial.h:94:5: note: SoftwareSerial::SoftwareSerial(int8_t, int8_t, bool)

     SoftwareSerial(int8_t rxPin, int8_t txPin = -1, bool invert = false);

     ^

C:\Users\Chema\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\SoftwareSerial\src/SoftwareSerial.h:94:5: note:   candidate expects 3 arguments, 4 provided

C:\Users\Chema\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\SoftwareSerial\src/SoftwareSerial.h:90:5: note: SoftwareSerial::SoftwareSerial()

     SoftwareSerial();

     ^

C:\Users\Chema\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\SoftwareSerial\src/SoftwareSerial.h:90:5: note:   candidate expects 0 arguments, 4 provided

exit status 1
no matching function for call to 'SoftwareSerial::SoftwareSerial(int&, int&, bool, const int&)'

```

djares01, 2020-03-26:
If you change your board manager model version to 2.4.2 it works

jantenhove, 2020-03-30:
I am currently testing a fix for this issue.

commit da44ff3c (2020-03-31): SoftwareSerial fix for ESP boards > 2.4.2. Fixes #23
closed 2020-03-31 by jantenhove