jantenhove/GoodWeLogger #24: Crashes after running for 1-2 hours

rispie, 2020-03-21:
At first it seem to work fine, however I noticed now that several times that it crashes. I'm using a NodeMCU v2: https://opencircuit.nl/Product/NodeMcu-v2-Lua-ESP-12E-WIFI-Development-Board and the board NodeMCU v1 (ESP-12E Module) and ESP8266 2.4.2. (Because it doesn't seem to compile on version > 2.4.2).

Do you maybe have any suggestions what could be causing this?

`12:38:23.923 -> CRC received: 0xA 0xFE , calculated CRC: 0xA 0xFE .
12:38:23.923 -> CRC match.
12:38:24.776 -> Sending data to inverter(s): 0xAA 0x55 0xAB 0x1 0x1 0x1 0x0 CRC high/low: 0x1 0xAD .
12:38:24.898 -> Parsing incoming data with length: 0x43 . 0xAA 0x55 0x1 0xAB 0x1 0x81 0x3C 0x9 0x48 0x8 0x17 0x0 0x3E 0x0 0x40 0x9 0x77 0x0 0x73 0x13 0x87 0xA 0x86 0x0 0x1 0x1 0x4C 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0x39 0x0 0x0 0x4 0xC2 0x3 0xF2 0x0 0x0 0x0 0xC8 0x9 0x60 0xE 0x8F 0x0 0x14 0x0 0x27 0x1 0x3 0xF 0x3 0x24 0x1D 0x0 0x0 0x0 0x0 0x0 0x8 0x0 0x0 0xA 0x36 .
12:38:24.898 -> CRC received: 0xA 0x36 , calculated CRC: 0xA 0x36 .
12:38:24.946 -> CRC match.
12:38:25.800 -> Sending data to inverter(s): 0xAA 0x55 0xAB 0x1 0x1 0x1 0x0 CRC high/low: 0x1 0xAD .
12:38:25.882 -> Parsing incoming data with length: 0x43 . 0xAA 0x55 0x1 0xAB 0x1 0x81 0x3C 0x9 0x38 0x7 0xFF 0x0 0x40 0x0 0x3F 0x9 0x76 0x0 0x73 0x13 0x87 0xA 0x8A 0x0 0x1 0x1 0x4D 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0x39 0x0 0x0 0x4 0xC2 0x3 0xF2 0x0 0x0 0x0 0xC8 0x9 0x60 0xE 0x8E 0x0 0x14 0x0 0x27 0x1 0x3 0xF 0x3 0x24 0x1F 0x0 0x0 0x0 0x0 0x0 0x8 0x0 0x0 0xB 0x13 .
12:38:25.923 -> CRC received: 0xB 0x13 , calculated CRC: 0xB 0x13 .
12:38:25.923 -> CRC match.
12:38:26.795 -> Sending data to inverter(s): 0xAA 0x55 0xAB 0x1 0x1 0x1 0x0 CRC high/low: 0x1 0xAD .
12:38:26.905 -> Parsing incoming data with length: 0x43 . 0xAA 0x55 0x1 0xAB 0x1 0x81 0x3C 0x9 0x5D 0x8 0x3 0x0 0x3E 0x0 0x41 0x9 0x78 0x0 0x73 0x13 0x86 0xA 0x8B 0x0 0x1 0x1 0x4D 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0x39 0x0 0x0 0x4 0xC2 0x3 0xF2 0x0 0x0 0x0 0xC8 0x9 0x60 0xE 0x8A 0x0 0x14 0x0 0x27 0x1 0x3 0xF 0x3 0x24 0x21 0x0 0x0 0x0 0x0 0x0 0x8 0x0 0x0 0xA 0x3D .
12:38:26.956 -> CRC received: 0xA 0x3D , calculated CRC: 0xA 0x3D .
12:38:26.956 -> CRC match.
12:38:27.826 -> Sending data to inverter(s): 
12:38:35.049 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
12:38:35.095 -> 
12:38:35.095 -> wdt reset
12:38:35.095 -> load 0x4010f000, len 1384, room 16 
12:38:35.095 -> tail 8
12:38:35.095 -> chksum 0x2d
12:38:35.095 -> csum 0x2d
12:38:35.095 -> vbb28d4a3
12:38:35.095 -> ~ld
12:38:35.197 -> Booting
12:38:35.249 -> Connecting to WIFI...`

Another crash:
`
12:56:55.307 -> Parsing incoming data with length: 0x43 . 0xAA 0x55 0x1 0xAB 0x1 0x81 0x3C 0x9 0x79 0x8 0x70 0x0 0x45 0x0 0x43 0x9 0x7C 0x0 0x7E 0x13 0x88 0xB 0xBB 0x0 0x1 0x1 0x5E 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0x41 0x0 0x0 0x4 0xC3 0x3 0xF2 0x0 0x0 0x0 0xC8 0x9 0x60 0xE 0x94 0x0 0x14 0x0 0x2F 0x1 0x3 0xF 0x3 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x8 0x0 0x0 0xB 0x2F .
12:56:55.307 -> CRC received: 0xB 0x2F , calculated CRC: 0xB 0x2F .
12:56:55.307 -> CRC match.
12:56:56.083 -> Sending discovery
12:56:56.083 -> Sending data to inverter(s): 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
12:56:57.097 -> Sending data to inverter(s): 0xAA 0x55 0xAB 0x1 0x1 0x1 0x0 CRC high/low: 0x1 0xAD .
12:57:04.297 -> 
12:57:04.297 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
12:57:04.297 -> 
12:57:04.297 -> wdt reset
12:57:04.297 -> load 0x4010f000, len 1384, room 16 
12:57:04.297 -> tail 8
12:57:04.297 -> chksum 0x2d
12:57:04.297 -> csum 0x2d
12:57:04.297 -> vbb28d4a3
12:57:04.297 -> ~ld
12:57:05.318 -> Booting
12:57:05.318 -> Connecting to WIFI...
12:57:05.451 -> ....Ready
12:57:09.263 -> IP address: 
`

jantenhove, 2020-03-30:
Thanks for the detailed logs.  It looks like the watchdog timer resets the ESP. 
I am currently testing a version that compiles with the latest ESP boards version. Maybe this fixes the issue. If not, I'll add some extra logging to catch this error.

jantenhove, 2020-03-31:
Too bad. The latest boards version did not fix this issue.  (commit da44ff3)
I will add extra (remote) logging too see what causes the wdt reset.

jantenhove, 2020-04-13:
Adding extra yields+using a clock speed of 160Mhz seems to fix this problem. Howevers problem #16 still persists.