jantenhove/GoodWeLogger #28: GoodWe XS-serie UART RS-232

popoviciri, 2020-05-21:
I've been trying for a couple of days to use this code with my small XS serie inverter.
The wifi module uses a [HF-LPB100](https://adaptivemodules.com/wireless-modules/hf-lpb100-wi-fi-module/) chip, which according to the [user manual](http://www.adaptivemodules.com/assets/Wi-Fi/HF-LPB100-Wi-Fi-Module-User-Manual.pdf) has a RS-232 UART interface. The USB3 pins are connected as show in the last column, here next to the [standard USB3 configuration](https://www.aggsoft.com/usb-pinout-cable/usb3.htm):

|stdUSB Pin|stdUSB Name|stdUSB Direction|stdUSB Color|stdUSB Description|Goodwe XS choice|
|-|-|-|-|-|-|
|1|VBUS||red|+5 V power|GROUND|
|2|D-|«—»|white|USB 2.0 Data -|not used|
|3|D+|«—»|green|USB 2.0 Data +|not used|
|4|GND||black|Ground|+5 V power|
|5|StdA_SSRX-|«—|purple|SuperSpeed receiver|not used|
|6|StdA_SSRX+|«—|orange|SuperSpeed receiver|UART Rx|
|7|GND_DRAIN||ground|Ground|UART Tx|
|8|StdA_SSTX-|—»|blue|SuperSpeed transmitter|UART Rx|
|9|StdA_SSTX+|—»|yellow|SuperSpeed transmitter|RESET|

**So it looks like a USB3.0 but has nothing to do with it!**
This is how I found the pins:
[image]

The comparison with the standard USB connectors:
[image]

**One should not stick a standard USB device in there. Will likely break it since the standard V+ and GND are reversed.**
I did cut a USB cable and connected only the pins I'm interested in Rx and Tx to a ESP8266 Huzzah. I can also power the Huzzah from the reversed pins 1 and 4 and connects fine to my wifi and subsequently mqtt broker.

The serial monitor shows a bunch of these lines:
```
18:01:02.855 -> Sending discovery
18:01:02.855 -> Sending data to inverter(s).Sent data to inverter(s):
18:01:02.855 -> 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
18:01:02.855 -> Parsing incoming data with length: 0x7 . 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 0x2 0x29 .
18:01:02.855 -> CRC received: 0x2 0x29 , calculated CRC: 0x2 0x29 .
18:01:02.908 -> CRC match.
18:01:02.908 -> MQTT send status: 1
```
Obviously this is not right. Incoming data from the inverter is identical with sent packet. So, is it safe to assume that the XS series uses a different protocol than the one this application is based on? How's the data coming from the inverter supposed to look like? When sniffing the port I get only nonsense in the terminal view at all baudrates I can set. Sniffing the wifi module shows nothing in the terminal.
I know this is supposed to be an issue tracker but the repo was found by multiple search engines in relation to the subject, so I post this here.
Goes without saying that I do not encourage anyone to try this. I'm just looking for the obvious detail I'm missing. Thanks in advance for any feedback!
Cheers!

No13, 2020-05-25:
I was looking to build something for my 1500-XS. The HF-LPB100 seems to 'forget' the AP SSID every time the sun goes down... (Why is disabling this AP feature not an option after setting up the STA mode!)

The output above is generated by the ESP8266 (based on code from this repository) when connected to the "USB" interface of the inverter? Without the HF-LPB100? 

I haven't gotten to hardware hacking yet and I was first trying to decode the TCP protocol. After all, after replacing the default wifi module it would be nice to update the SEMS portal from my own module ;)

No13, 2020-05-25:
Update: I connected a Logic Analyser and found the baud rate between the inverter and wifi module is 9600 baud. 

Also I see the same messages on the serial bus as seen on the wireshark dump (1 minute interval)

```
00000000  50 4f 53 54 47 57 00 00  00 f9 01 04 00 00 [       POSTGW.. ......XX
00000010  Serial number here (bytes 14 to 29)      ] 14 01   XXXXXXXX XXXXXX..
00000020  0e 12 34 2a 00 00 00 00  00 00 00 00 00 00 14 01   ..4*.... ........
00000030  0e 12 34 2a e3 e0 c4 fa  5c 38 3a d9 6a bc 80 e4   ..4*.... \8:.j...
00000040  1b 7f 30 bb 3e ba db eb  fd fd 6a f3 8e c6 fe 98   ..0.>... ..j.....
00000050  46 12 40 e7 f9 1a 10 63  da 73 e2 67 72 14 67 89   F.@....c .s.gr.g.
00000060  f7 3b 65 ce 0e ec 24 7a  19 b4 45 ca 0d b1 79 4e   .;e...$z ..E...yN
00000070  32 0c da a8 3f 1d 3b 05  84 7e 79 9a e9 36 69 a1   2...?.;. .~y..6i.
00000080  9a 88 d9 de 84 a1 25 92  07 b8 cf e7 49 26 0e b6   ......%. ....I&..
00000090  e2 0e 0e e9 55 aa 3f fb  46 e4 e0 1e 22 22 7b c3   ....U.?. F...""{.
000000A0  ef 99 c5 fe b2 97 e2 0c  ab b3 b9 cb bd 79 d8 7c   ........ .....y.|
000000B0  d0 13 57 62 5b 1f f5 5c  78 f0 5a 75 84 06 58 d3   ..Wb[..\ x.Zu..X.
000000C0  4f e5 eb da 9e 32 e5 b1  99 58 8b d1 24 26 f6 87   O....2.. .X..$&..
000000D0  cd b0 df 14 f6 d6 3d 3d  da 28 2c c1 71 ed 04 67   ......== .(,.q..g
000000E0  bf 2d 65 1b e8 f2 d3 f9  75 82 0c fa c3 9c e8 40   .-e..... u......@
000000F0  e7 00 92 45 82 b5 32 08  b3 28 5e 1b fe d6 07 36   ...E..2. .(^....6
00000100  2f a8 12 10 38 8d                                  /...8.```

This HF-LPB100 seems to be a serial to TCP bridge only.
So far I found the following 'attributes' in this stream:

Messages from inverter:
- Start with POSTGW (ASCII) 0x00 0x00 0x00 (bytes 0 - 8)
- Byte 9 is the length of the actual data (0xf9 = 249)
- Bytes 10 - 13 seem to contain static data: 0x01 0x04 0x00 0x00
- Bytes 14 - 29 contain the inverters serial number (ASCII)
- Last two bytes see to be CRC check
```
00000000  47 57 00 00 00 31 01 04  [ Serial number           GW...1.. XXXXXXXX
00000010                         ] 1e 35 0f 0e 05 14 00 00   XXXXXXXX .5......
00000020  00 00 00 00 00 00 00 00  35 74 8a e8 33 0f 12 56   ........ 5t..3..V
00000030  c7 09 18 1d 7e c4 db ea  af c4                     ....~... ..
```
Messages from server (or to inverter):
- Start with GW (ASCII) 0x00 0x00 0x00 (bytes 0 - 4)
- Byte 5 is the length to the actual data
- Bytes 6 - 7 seem to contain static data: 0x01 0x04
- Bytes 8 - 23 contain the serial number (ASCII)
- Last two bytes contain the CRC again

No13, 2020-05-25:
After rebooting the wifi modile I'm seeing AT commands,
Some checking of the wifi link and finally:
AT+ENTM: Set module into transparent transmission mode

So yes, the wifi module is a simple serial to IP bridge

jantenhove, 2020-05-25:
Wow, great work both of you. Really detailed analysis.
The inverters do indeed use a different protocol as their bigger brothers, but the good new is that they don't use any encryption since the serial number is visible in the data stream. 
Do you now the day total for the captured data? If we're lucky they use the same float to byte packing as the original; `float(((unsigned short)bt[0] << 8) | bt[1]) / factor;`
@No13  Do I understand it correctly that, when the serial communication is deciphered, the Wifi module can be replaced by an ESP?

No13, 2020-05-25:
The final 2 bytes are indeed CRC and the algorithm turns out to be "CRC-16/MODBUS".

I am fairly certain the voltage level is 3.3v and it's a regular 9600 baud serial interface. If we record the dialog between the original wireless module we should be able to simulate the wifi dongle using an ESP (Only a few AT commands). After the AT+ENTM command it's a simple serial to TCP bridge, no problem for the ESP.

I will check if your provided code results to recognisable data. 

This is the AT dialog in my example (MYSSID and MYBSSID replaced). RX is the inverter and TX is the wireless module
```
RX: +++
TX: a
RX: a
TX: +ok[0D][0A][0D][0A]
RX: AT+WSLK[0D][0A]
TX: AT+WSLK[0D][0A][0D]
TX: +ok=MYSSID(MYBSSID)[0D][0A][0D][0A]
RX: AT+WSLQ[0D][0A]
TX: AT+WSLQ[0D][0A][0D]
TX: +ok=Good, 100%[0D][0A][0D][0A]
RX: AT+ENTM[0D][0A]
TX: AT+ENTM[0D][0A][0D]
TX: +ok[0D][0A][0D][0A]
RX: POSTGW[00][00][00][F9] [..]
```

No13, 2020-05-25:
The total kW should be fairly static but I'm getting nowhere

No13, 2020-06-08:
I made a datalogger to write the stream to disk over time, but the data after the SN does seem to be encrypted or compressed somehow (high entropy)

popoviciri, 2020-06-10:
I did follow with interest your dialog here, however I'm afraid I am not knowledgeable enough to further contribute. In the meanwhile I moved on to setup a [kWh meter](https://www.elektramat.nl/emat-dinrail-kwh-meter-45a-1-fase-digitaal-mid-gekeurd/) pulsecounter in esphome with an integration sensor in home assistant.
I don't like to leave lingering open issues around, so feel free to close this if becomes stale.
Cheers!

ThinkPadNL, 2021-02-04:
@No13 I found [this](https://brnrd.eu/misc/2019-03-23/killing-the-internet-of-shit.html) blog from someone who seems to have figured out (part of) the data. Maybe that can help you?

ThinkPadNL, 2021-02-06:
I now also have a GW3000-XS (i got tired of the relay clicking of my 3000NS at twilight). I have the same module as the pictures in the first message. I made a dump of the SPI chip (hidden under the metal shield). These are the AT-commands that i found in the file:
`AT+ASWD: Set/Query WiFi configuration code. 
AT+E: Echo ON/Off, to turn on/off command line echo function. 
AT+ENTM: Goto Through MOde.
AT+NETP: Set/Get the Net Protocol Parameters.
AT+PSPAR: Set/Query power save parameters.
AT+MSLP: Set/Query deep sleep mode parameters.
AT+MSOPT: Set/Query wake up mode parameters.
AT+TSPAR: Set/Query timeout/wake up parameters.
AT+TXPWR: Set/Query the TX power.
AT+UARTF: Enable/disable UART AutoFrame function.
AT+UARTFT: Set/Get time of UART AutoFrame.
AT+UARTFL: Set/Get frame length of UART AutoFrame.
AT+UARTTE: Set/Query UART free-frame triggerf time between two byte.
AT+PING: General PING command.
AT+WMODE: Set/Get the WIFI Operation Mode (AP or STA).
AT+WSLK: Get Link Status of the Module (Only for STA Mode).
AT+WSLQ: Get Link Quality of the Module (Only for STA Mode).
AT+WSCAN: Get The AP site Survey (only for STA Mode).
AT+TCPLK: Get The state of TCP link.
AT+TCPTO: Set/Get TCP time out.
AT+TCPDIS: Connect/Dis-connect the TCP Client link
AT+MAXSK: Set/Get MAX num of TCP socket (1~32)
AT+RECV: Recv data from WIFI
AT+SEND: Send data to WIFI
AT+DISPS: Disable power saving mode of WIFI
AT+WEBU: Set/Get the Login Parameters of WEB page.
AT+WEBVER: Get WEB version.
AT+WSDNS: Set/Get the DNS Server address.
AT+WADMN: Set/Get the domain name of WEB page.
AT+WEBSWITCH: Set/Get the parameters of WEB page.
AT+PLANG: Set/Get the language of WEB page.
AT+UPURL: Set/Get the path of remote upgrade.
AT+UPFILE: Set/Get the file name of config file for remote upgrade.
AT+UPST: Start the remote upgrade.
AT+UPWEB: Start the remote upgrade webpages.
AT+UPCFG: Start the remote upgrade default setting.
AT+UPAUTO: Start the remote upgrade by config file.
AT+LOGSW: Enable/Disable upload logs.
AT+LOGPORT: Set/Get the UDP port for upload logs.
AT+SOCKB: Set/Get Parameters of socket_b.
AT+TCPLKB: Get The state of TCP_B link.
AT+TCPTOB: Set/Get TCP_B time out.
AT+TCPDISB: Connect/Dis-connect the TCP_B Client link.
AT+RCVB: Recv data from socket_b
AT+SNDB: Send data to socket_b
AT+RELD: Reload the default setting and reboot.
AT+SLPEN: Put on/off the GPIO7.
AT+RLDEN: Put on/off the GPIO45.
AT+Z: Reset the Module.
AT+MID: Get The Module ID.
AT+VER: Get application version.
AT+H: Help.`

@No13 Let me know if i can help you with anything.

ThinkPadNL, 2021-02-06:
@No13 See https://gathering.tweakers.net/forum/list_message/66034946#66034946 for a script that a user on Tweakers.net is using to intercept the data. @popoviciri this might also be useful for you.

Don't know if this repository is still the right place to talk about this, as the way of handling the data is very different from the RS485-method that the sketch was written for.

ThinkPadNL, 2021-02-08:
Strange thing is by the way that i have the option 'Set modbus address' in the menu of the inverter. Could the protocol we are looking at be modbus by any chance?

I also had a look at the connectors of the inverter. In the manual i see something about the 'DRED' connector. This is for remote control of the inverter by energy companies (shutting down if there is too much production for example). I have the impression when looking at the manual, that if your inverter is RS485 capable, this connector is also used for that purpose. Unfortunately only two pins on this connector are populated in my case, so no RS485.

ThinkPadNL, 2021-02-10:
@popoviciri When looking at the photos of the module (and my own), i noticed that the TX of the UART is connected to two pins of the USB-connector. I am thinking of why this is done, maybe so the inverter can see if there is something connected?

I have now turned my attention to see if there is a RS485 connection anywhere. The TCP messages seem uncrackable for now.

ThinkPadNL, 2021-02-21:
@popoviciri I @No13 found some very interesting info! On the Goodwe website there is a download called [
_GW_Firmware & PC Software & Instruction All-in-one for CEI 021_
](https://www.goodwe.com/Public/Uploads/sersups/GW_Firmware%20&%20PC%20Software%20&%20Instruction%20All-in-one%20for%20CEI%20021.rar). I have downloaded it and it contains firmware files, update tools and instructions for multiple types of inverters and also the XS-series! 
[Local Upgrade for XS Method.pdf](https://github.com/jantenhove/GoodWeLogger/files/6018360/Local.Upgrade.for.XS.Method.pdf)
I attached the documents regarding the instructions for the XS.

As you can see they use a cable they call 'TTL-485-USB cable' to update the inverter with the utility 'DataSend for COM'.

When looking at the instructions for a SMT-type inverter in the download package, they have included a manual on how to construct an updating cable for the SMT-type inverter. It uses only two wires and works with the same program (DataSend for COM). This makes me think that there actually is a RS485 port in our XS-series, but not as a separate connector, but through the USB3 connector. 

Now only to see if that is really the case and how to find out the pinout.. When looking at your schematics, i see that pins 2 & 3 and 5&6 are not used by the original wifi stick. Would it really be that simple that the RS485 lives on one of these.... I will see if i can measure/try something this week.

popoviciri, 2021-02-22:
Hi @ThinkPadNL, great finds! I didn't think it'll be a RS485 port in there as well. If there is, then must be on pin2 and pin3 for a standard USB2 connector. When I catch a moment from work, I'll connect to those and see if I get anything. Anyway, I believe that the 'TTL-485-USB cable' you mention above it's likely a GoodWe internal cable since the Vs and GND pins on the inverter are reversed. I did powered an ESP8266 via that source or maybe there is no ground needed for the connection and just the two data pins? I use this [TTL to RS485](https://www.benselectronics.nl/uart-ttl-to-rs485-converter.html), board which has a GND pad as well so I assume it is needed. By the way, I have two of these so if you need one, ping me on tweakers, and I'll send it to you.
Cheers!

ThinkPadNL, 2021-02-22:
@popoviciri I was able to find a USB3 cable. I will conduct some measurements tomorrow (inverter is off because it is dark now).

Any tips where to look for? When looking at your schematics, i see that TX is not only connected to pin 8, but also to pin 6 (why???). So that leaves us with 3 free pins: 2, 3, 5 i guess? Just measure between GND (pin 1 on USB3 connector in this case) and those three? Any other suggestions?

With my previous inverter (NS-series) it only used the two A & B pins for RS485, no GND was needed. 

Thanks for the offer for the converter PCB, but it's not needed. 👍 I have that exact PCB lying around. Have used it for some years with my 3000NS (which i swapped with a 3000XS because that one hass much less clicking relay noise at twilight/bad weather).

ThinkPadNL, 2021-02-22:
I remembered that when contacting Goodwe about options for local data retrieval on the XS-series, they send me a PDF with information about the Modbus RTU protocol, see [this document](https://github.com/jantenhove/GoodWeLogger/files/6024769/Goodwe_Modbus_PROTOCOL_Grid_English-ningzenkun20200527.1.pdf) i attached.
When i asked which connector to use, they pointed me to the RS485 connector, which according to the user manual should be available on the big round connector. But my inverter only has two pins occupied for the DRED functionality. The other holes in the connector are empty.

That is the physical part, but it makes me think that this inverter uses a different protocol (Modbus) than what the code in this repository was written for (some proprietary protocol with handshake and such). However i have zero experience with modbus and the protocol PDF is not making it more clear to be honest. I can also imagine that we first need to send a 'please give me data' command to the inverter before it replies. So i am wondering if my voltmeter measurements will give me any useful info.

Any ideas about this @popoviciri and @No13 ?

ThinkPadNL, 2021-02-23:
A new, sunny, day. As i work from home i had some time to experiment. 
**TL;DR**: No RS485/Modbus found yet 👎 

I started with measuring the pins on the USB3 connector. One lead of the multimeter i clipped to pin 1 (which is used as GND by the XS). Then probed all remaining pins:

| Pin combination  | Voltage | Purpose
| -| - |-|
|1-2 |3,24V|?|
|1-3 |0,01V|?|
|1-4 |4,99V|Vin|
|1-5 |0,00V|?|
|1-6 |0,01V|? / bridged with pin 8 on wifi-module|
|1-7 |3,28V|Rx|
|1-8 |0,12V|Tx|
|1-9 |3,28V|Reset|

I then attached a RS485-TTL PCB to a USB-Serial converter. I then downloaded [Modbus Poll](https://www.modbustools.com/modbus_poll.html) and configured it (9600 8N1), entered the modbus settings as per the Goodwe PDF (slave address 247, function 03). The interesting information starts from register 200 up to 237 so i used starting address 200, quantity 37. 
I see the TX light on the USB-Serial and RS485-TTL flicker, so the sending part is ok, But the 'RX' light never turns on.

 I tried all possible combinations of wires. Then i also did the same but with the ESP connected to the RS485-TTL and with the sketch from this repository. Still no data (or RX light) from inverter.

ThinkPadNL, 2021-02-23:
Goodwe replied _that the cable is hard to find in the Netherlands. It has to be specially made, they only have two of it available.
Also there are two types of connections RS485 and TTL._
He told me that based on the serial number of my inverter, it doesn't have RS485, only TTL. When i asked if they have a protocol description for TTL he said that only the RS485 protocol is publicly available.

popoviciri, 2021-02-23:
Pin 2 and 3 from the USB are connected to these red + black wires plugged into the board.
[image]
The bigger cable is connected like so:
[image]
Probably not relevant, but well.
I got the same results as you, @ThinkPadNL, hoping the port offers parallel communication with both RS485 and TTL, although thinking about it sounds unlikely. I measure the same voltage as you, by the way. Anyway, there are sunny days the coming so I'll do some more testing.

ThinkPadNL, 2021-02-23:
Whoaaaa, those are some great pictures! 🙌 Especially because i had the same thoughts a few weeks ago, to open up the inverter. But i was held back by the 'warranty void' sticker 😇  I already tried removing it very carefully with a hairdryer and a sharp knife, but after trying only 0.1mm i already saw i couldn't remove it without damaging it, so i stopped. You were not impressed by it and ripped it off anyway? 🤣 

The amount of wires in the bigger white connector matches exactly the pins that you pointed out on the wifi-stick, so no extra pins/functionality there. Interesting to see that separate red/black connector. You would really think it could offer RS485 on those, but as we have both tried it doesn't work. If i understand the manual/Goodwe support correctly there are two types of inverters, where one has RS485 and one has not (ours). I would assume that for the RS485 they connect some extra wires/PCB-module somewhere and use different firmware on it.

What about the separate PCB that the purple and yellow wire goes to? Does it have any chips/additional connectors on it?

And maybe you can make some more sharp pictures of the main PCB and internals of the inverter? Maybe we can identify some more interesting things. And also just to feed my curiosity 😄

popoviciri, 2021-02-24:
Right! I opened the box months ago, when I was first trying to log it to my home-assistant. I only have the 700W version which is not really worth a second thought. I found that the rubber seal in the cover was torn and not sealing properly so by opening it up, likely I avoided a warranty claim. It is installed in a barn which can get pretty humid.

So there is some contradicting info there: on one hand the RS485 resides with the USB connection (most likely pin 2 and 3) implied by the special USB cable. On the other hand, according to the manual and feedback from GoodWe people the other connector is responsible for it. So yeah... I do wonder what are those black and white cables for.

The remaining pins on the other 8-pin connectors should probably go to the small PCB you asked for. It is soldered to the mainboard so these are the best shots I could get:
[image]

There is nothing written on that SOIC14 chip but is fair to assume is a RS485 module. I wonder if there is firmware support for RS485, since it's not even wired. But I'll definitely solder a pin-out and get the harness out of the box to try it out. Don't wanna power it up without the display which is on the cover. Should be easy to find which two pins are needed:
[image]

Here are more pics:
[image]
[image]
[image]

A well made inverter, in my opinion.

Cheers!

ThinkPadNL, 2021-02-24:
Very interesting, thanks for making these! @No13 and i both did a search with 'strings' utility on the firmware files. In the `master.out` file for the XS i got some strings that contain 'RS485'. 

I had a look in the datasheet for the `MAX48x` (a chip which is often used for RS485 i understand). In the datasheet there is a SOIC14 member of the family, the MAX489 and MAX491. Maybe that can help you in finding the pinout.

I'm not sure about the protocol though. It could be the protocol (proprietary) that this sketch uses, but also modbus (as the display has a menu option to set the Modbuss address). But if i remember correctly my old GW3000-NS also had that modbus option in the menu, but still used the proprietary protocol.

Keep us updated. I would be a decision struggle for me and others to also use it, as it requires to take off the lid 😆

popoviciri, 2021-02-24:
Thanks for the hints. I tried to check connections between pin 12 and 13 from the SOIC14 (A & B for MAX chips) but they're all pulled up or down and there is always something resistance to measure between those and any of the pins on the small PCB.
I got a cable out of the box, so I can test more.
[image]

So far I mapped the pins the same way, against GND.

Pin | Voltage | Purpose
-- | -- | --
1 | 0.1V | ?
2 | 0.1V | same as 1
3 | 0.1V | same as 1 & 2
4 | 4.85V | ?
5 | 4.85V | same as 4
6 | 0.04V | ?
7 | 0V | GND
8 | 0V | Not connected

I assumed that I'd get a signal from either of 4-5 with either of 1-2-3 connections, but no joy. Yet! I'll try again in the weekend and report back here. I never connect the WiFi dongle. Don't even know where it is anymore.

I got some closeup of the chips:
[image]
[image]

I though I got a better photo of the CPU but if you squint your eye a little, you can read it.
Cheers!

ThinkPadNL, 2021-02-27:
@popoviciri Keep us updated. I am a bit out of options for now unfortunately ☹️ The [reverse engineering of the TCP-datastream](https://gathering.tweakers.net/forum/list_messages/2052604) also seems quite difficult and doesn't have much progress. 

I get the feeling that it may not be possible at all to extract data from it locally.

popoviciri, 2021-02-27:
Hi @ThinkPadNL, after many unsuccessful trials with TTL to RS485 board, USB-serial converters and ESP8266 with the sketch from this repository, I decided to order a Logic Analyzer and inspect the signal that way. Will be delivered today so I'll be able to try it tomorrow.

ThinkPadNL, 2021-02-27:
Did you also try with Modbus? I hope a logic analyzer brings us new insights. Could still be difficult though, as the inverter could stay silent until it is asked to present his data.

popoviciri, 2021-02-28:
Hi @ThinkPadNL,
I was unsuccessful getting any data from the alleged RS485 interface. I tried pin 2 and 3 from the USB connection as well as various combinations from the internal PCB. I also hooked up an esp8266 running the sketch from this repository to the said pins in all possible combinations via the TTL to RS485 board and no joy. The Logic Analyzer pics up the esp8266 sending out the `0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0` data CRC `0x2 0x29` but the inverter says nothing in return.
To make sure I did connect the analyzer correctly, I also dug out the WiFi module and hooked it to the inverter. I found that months ago I already soldered jumper wires to Rx and Tx pins so sniffing the traffic was easy. Captured the traffic as one should and basically replicated @no13's findings, so that works fine.
Rebooted the inverter after inserting and removing the wifi and allowed sufficient time for data flow.
Same for Modbus. I'm afraid this is a dead end ...

ThinkPadNL, 2021-02-28:
To be honest, I was already afraid of that 😞 

Still makes me wonder how the update procedure works though, especially which pins they use. It could be the TX/RX pins that the wifi module uses (and thus simple TTL serial). But why then the separate black/red cable... 

As can be seen [here](https://github.com/jantenhove/GoodWeLogger/issues/28#issuecomment-633685012) there is some 'negotiation' between inverter and wifi-module. Maybe they send a special command which puts the inverter into update mode. Maybe we can try to sniff the serial output during startup of the inverter (and with wifi removed?), maybe it prints some diagnostic info? Or just send a `?` to see if it returns something?

bwired-nl, 2021-03-01:
Hi Guys
I was just following this and saw that the inverter Wifi menu presented by ThinkpadNL is exactly the same as my omniksol converter!
So could it be that the using the exact configuration and firmware?
With the omniksol you have to negotiate also to get the values
Please check below, perhaps a help.......
 
https://github.com/cyberjunky/wemosomnikserialsensor/blob/master/WemosOmnikSerialSensor.ino

ThinkPadNL, 2021-03-01:
Hi @bwired-nl ! (_familiar nickname, long time ago from domoticaforum.eu_ 😄 ),

I have looked at that code and see it uses port 8899 to communicate with the inverter. I have already tried that (could be that i have not posted that here but on Tweakers.net). There is similar code that uses this port 8899: [here](https://github.com/mletenay/home-assistant-goodwe-inverter) and [here](https://github.com/XtheOne/Inverter-Data-Logger). I've tried them both and got no response at all from my inverter.

But thanks anyway for thinking along 👍

gracenho, 2021-03-06:
Hi Everyone,
I also have the same dilemma, i have one gw3000xs, but i can read all data from rs485 modbus, write powerfactor but i cant write power limiter for zero injection, i find the modbus protocol from the version DT, and all match.
I read on comments may be only work with wifi or rs485 not both. 

Some Info:

4. Inverter Address: Can be assigned from 1~247. 247 is factory default assignment.
5. Communication baudrate: 9600bps
6. Function Code:
03H: Read Operation (NOTE: can read more than one registers at once)
10H: Write Operation (NOTE: Only support write single register at once)
7. CRC Code Verification
7.1 CRC multinomial：X16+X12+X5+1
7.2 CRC verification covers first byte to the last byte before CRC data.
7.3 Refer to chapter 11 to implement CRC verification

Power Adjust 1-100% adress modbus 0100
Power Factor  1-100 adress modbus 0101 1-10 as 0.99-0.9 lagging 90-100 as leading 0.90-1

https://github.com/MiG-41/Modbus-GoodWe-DT
my project is manage power draw with a powermeter DDS238-2 ZN-S very mutch cheap then gm1000...

ThinkPadNL, 2021-03-07:
Hi @gracenho,

Which connector / cable do you use for Modbus communication with the inverter? Was your inverter also sold with wifi?

And which country are you located in? It could be that there are differences in specifications in each country. Everyone in this thread (including myself) is from the Netherlands I thought.

gracenho, 2021-03-07:
Hi @ThinkPadNL,

I use this 
[image]
today i finally can adjust all parameters, my problem was the wifi stick ...
At this moment i use a esp8266 with ttl to rs485 to mqtt and on mqtt i manage the inverter, and a usb to rs485 its a usb to ttl and ttl to rs485... for debug.
Im from Portugal.
I can Upload a pdf from goodwe on git hub? its all data from register map off modbus goodwe.

ThinkPadNL, 2021-03-07:
The inverter that i (and also the others on this thread) have, only have pins 4 & 5 populated in the connector. The other pins (for RS485) are empty, so there is physically no possibility for us to use RS485.

No need for the PDF, i already got it from Goodwe 👍

gracenho, 2021-03-07:
Yes I have a com board...
[image]

[image]

ThinkPadNL, 2021-03-07:
Thanks for the pictures, maybe this gives @popoviciri new insights in a possible way to get RS485 on our inverters.

Badwater-Frank, 2021-04-01:
My XS1500 (bought in Germany) is equipped with a 14k4 resistor across pin7+8 (beside the remote bridge cable 4+5), which I expected to be rs485. But so far no success getting data out of it :-(

aiolos, 2021-04-13:
I have two GoodWe XS inverters, a XS2000 and a XS2500. Bought last month in NL. They also have this resistor and a description of the RS-485 pins in the manual.
With the available modbus registers (which I found in this issue or at some of the related sources) I created a plugin for ESPEasy to read the values. It's working for a few days for me already. The plugin can be found here: https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/173 (it's not in ESPEasy yet, if someone wants a binary with the plugin integrated, let me know)

About the pinout: I used pin 3&7 for A and pin 6&8 for B. (The RS-485+ and RS-485- are a bit confusing in the manual). I removed the resistor, but it also worked with it.

Badwater-Frank, 2021-04-17:
My XS1500 (FW ver. 1.03.09) will is not giving any data on the plug (named 4. un user guide) between mains and WLAN (USB) :-(
Tried with two different RS485 adapters for PC and QModMaster SW; tried pin3/7 and 6/8 interchanged +/-, checked signals with scope ...
Might the availability of the RS485 be controlled by any other setting?
My inverter is set to "Germany" ....

Arut42, 2021-05-02:
@ThinkPadNL would it be possible to ubload the bin-file from the spi flash?
I destroied mine :(
Btw. you can donwload the flash with UART0 921.600Baud
Hit "Enter(CRLF)" while power up and you get into boot 
Here you can download the flash with "upload 0x18001000 0x20000" // Modem-x into file or hterm
0x18001000 is the User partition 0x20000 is the size it should be 2MB or maybe 1MB
Clean your privat data in 0x18007000 - 0x18009000

would be great
best regards

ThinkPadNL, 2021-05-03:
@Arut42 I already did that once, for someone who modified the .html pages (to hide his network password out of the open).
He reported back to me that he got his wifi working again, but it was sending out the SSID with part of **my** inverter's serialnumber in it. I don't really like that. So i will not upload my firmware, sorry.

Arut42, 2021-05-03:
@ThinkPadNL i can understand that.
i found this partition table can u verify that.
my problem is i found only the firmware on hi-flying but goodwe use custom firm.
so my inverter is not runing 
i only need the firmware part
#define SYSTEM_SECTOR_ADDRESS			0x18000000
#define SYSTEM_SECTOR_SIZE				(0x1000)
#define BOOT_ADDRESS					0x18001000
#define BOOT_SIZE						(0x3000)
#define BOOT_CONFIG						0x18004000 
#define SOC_CONFIG_ADDRESS				0x18005000
#define SECU_FLASH_ADDR					0x18006000
#define USERPAGE 						0x18007000
#define USERPAGE_BACKUP 				0x18008000
#define F_SETTING_ADDRESS       		0x18009000
#define USER_BIN_FILE_OFFSET			0x1800A000
#define USER_BIN_FILE_MQTT_OFFSET		0x1800AB30     //size(1100) test by ZN ----20190610  
#define USER_BIN_FILE_SIZE				4*1024
#define USER_BIN_FILE_BAK_OFFSET		0x1800B000

#define SYSTEM_CONFIG_SIZE				(0x8000)

#define SOFTWARE_CODE_ADDRESS			0x1800C000
#define MAX_SOFTWARE_CODE_SIZE			(800*1024)
#define UPGRADE_ADDRESS					0x18100000
#define MAX_UPGRADE_FILE_SIZE			(512*1024)
#define UPGRADE_ADDRESS_END				(UPGRADE_ADDRESS+MAX_UPGRADE_FILE_SIZE)

#define WEB_ADDRESS						0x18180000		//200k
#define WEB_ADDRESS_END					0x181B2000
#define WEB_SCAN_TMPBUF					0x0
#define HFUFLASH_SIZE					(160*1024)
#define HFUFLASH1_SIZE					(0*1024)
#define UFLASH_ADDRESS					0x180D4000
#define UFLASH1_ADDRESS					0x0
#define TEMP_FLASH_ADDRESS				0x181FF000

ThinkPadNL, 2021-05-03:
@Arut42 I don't know how i can extract only the firmware part. What i have is a `.bin` that i dumped (i thought with `flashrom` utility on a RPi).

ThinkPadNL, 2021-05-03:
@No13 @aiolos @popoviciri @Badwater-Frank @gracenho I have great news.
A user on 'Tweakers.net' found out that on XS-serie with newer firmware, the app 'SolarGo' from Goodwe can be used to retrieve data. On my 3000XS it didn't work at first (Error message "Your inverter is unsupported"), but the guy on Tweakers found in a datasheet somewhere that the ARM-firmware should be >=13 if you want to use Modbus.

I asked Goodwe to update my inverter (which was at v1.02.10). They first said it wasn't possible to do remotely and that it would cause issues (???). I then said i wanted to enable Modbus. They proceeded to update, to v1.52.10. SolarGo still wasn't working. I then asked them again to also update the ARM-firmware. This time my inverter went to v1.52.14 and SolarGo immediately worked!

The code from the user ('msatter') can be found here: https://gathering.tweakers.net/forum/list_message/67162456#67162456
It also works for me, but it feels a bit clunky for me. I prefer something like Node-RED.

Arut42, 2021-05-04:
@ThinkPadNL u can view it with an hex editor like https://hexed.it/
address 0x0 up to 0x3000 should be the bootloader (maybee an address offset of 0x1000)
system firmware should be at 0xC000 up to around 0x15 0000
and webpage should be at 0x18 0000

ThinkPadNL, 2021-05-07:
@Arut42 In the link you can find the firmware dump from my wifi-module (HF-LPB100). I searched for the (part of) the serial no of my inverter in the code and replaced it with '00000000'. Download it [here](https://www.dropbox.com/s/nvy22qos1cinpas/goodwe_HF_LPB100wifi_EN25QH16_sanitized.bin?dl=0). Let me know if it worked.

Arut42, 2021-05-18:
@ThinkPadNL thx for your work.
Looks like u have some different memory alignment. got a "Hard Faults Report"
Have to search more online, there is a second number from goodwe: HF-GDW-0004

Badwater-Frank, 2021-07-09:
Unfortunately the support replies after some time, that inverters (mine XS1500) produced before October 2020 do not have the RS485 feature :-(
My inverter produced in 2019 ... bad luck ...

ThinkPadNL, 2021-07-09:
@Badwater-Frank Have a look in this topic: https://gathering.tweakers.net/forum/list_message/67162456#67162456
Check if you can read data with the 'SolarGo' app. If that works you can use the scripts in that topic. If SolarGo is not working, ask Goodwe to update the 'ARM-firmware' (last two digits of version number displayed by the inverter) to xx.xx.13 or higher.

After they updated the ARM-firmware on my 3000XS i was able to read data through wifi using the SolarGo, and that opened possibilities to use other scripts to read the inverter data locally.

@popoviciri I think the original question is not so relevant anymore. Maybe it is time to close this issue now.

popoviciri, 2021-07-09:
Right! Thanks @ThinkPadNL for the info provided. This thread was fun to follow.
Happy tweaking! Cheers!

closed 2021-07-09 by popoviciri

HoyaTon123, 2023-09-28:
> I have two GoodWe XS inverters, a XS2000 and a XS2500. Bought last month in NL. They also have this resistor and a description of the RS-485 pins in the manual. With the available modbus registers (which I found in this issue or at some of the related sources) I created a plugin for ESPEasy to read the values. It's working for a few days for me already. The plugin can be found here: [letscontrolit/ESPEasyPluginPlayground#173](https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/173) (it's not in ESPEasy yet, if someone wants a binary with the plugin integrated, let me know)
> 
> About the pinout: I used pin 3&7 for A and pin 6&8 for B. (The RS-485+ and RS-485- are a bit confusing in the manual). I removed the resistor, but it also worked with it.

Hi there, I just bought a XS-3000 but without RS-485, I realized it become a optional choose, but it was too late.
But I would like to make sure it has a 485 pin just hidding inside the mother board? If so, I will open it up and try to connect it.