jantenhove/GoodWeLogger #32: No output from logger

ferdyh, 2020-12-23:
I've got a GW-3600D-NS, hooked it up and no output to MQTT except for goodwe = online.

Ouptut on the remote debug:
```
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
```
Any suggestions on how to fix? or what i did wrong?

davevleugel, 2021-01-12:
I have the same problem. I tested with the L0Lin and wemos node. It did work for 6 months or so until I changed my SSID in the code because of an other AP. I did change the AP back and updated my SSID again to the old one, but the same result as @ferdyh 

This is my output from the first sunlight when my inverter boots up (3600d-ns). I used the reboot command to try it again, same result:

```
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Parsing incoming data with length: 0x17 . 0xAA 0x55 0x7F 0xAB 0x0 0x80 0x10                         0x35 0x33 0x36 0x30 0x30 0x44 0x53 0x4E 0x32 0x30 0x31 0x57 0x30 0x30 0x35 0x31                         0x6 0x4C .
(D) CRC received: 0x6 0x4C , calculated CRC: 0x6 0x4C .
(D) CRC match.
(D) New inverter found. Current # registrations: 1
(D) SendAllocateRegisterAddress address: 1
(D) Sending data to inverter(s).Sent data to inverter(s):
reboot
*** Remote debug - over telnet - for ESP8266 (NodeMCU) - version 3.0.5
* Host name: GoodweLogger IP:192.168.1.46 Mac address:48:3F:DA:45:73:F4
* Free Heap RAM: 27096
* ESP SDK version: 2.2.2-dev(38a443e)
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter
    cpu80  -> ESP8266 CPU a 80MHz
    cpu160 -> ESP8266 CPU a 160MHz
    reset -> reset the ESP8266

****
* New features available:
* - Now you can debug in web browser too.
*   Please access: http://joaolopesf.net/remotedebugapp
* - Now you can add an simple software debuggger.
*   Please access: https://github.com/JoaoLopesF/RemoteDebugger
****

* Please type the command and press enter to execute.(? or h for this help)
***
(D) ===GoodWeLogger remote debug enabled===
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
v
* Debug: Command received: v
* Debug level set to Verbose
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Sending discovery
(D) Sending data to inverter(s).Sent data to inverter(s):
(D) 0xAA 0x55 0xAB 0x7F 0x0 0x0 0x0 CRC high/low: 0x2 0x29 .
(D) MQTT send status: 1
(D) Parsing incoming data with length: 0x17 . 0xAA 0x55 0x7F 0xAB 0x0 0x80 0x10 0x35 0x33 0x36 0x30 0x30 0x44 0x53 0x4E 0x32 0x30 0x31 0x57 0x30 0x30 0x35 0x31 0x6 0x4C .
(D) CRC received: 0x6 0x4C , calculated CRC: 0x6 0x4C .
(D) CRC match.
(D) New inverter found. Current # registrations: 1
(D) SendAllocateRegisterAddress address: 1
(D) Sending data to inverter(s).Sent data to inverter(s):

```

I also tryed diffrent CPU speed and flash size with same results. I hope someone has a solution to this.

Kind Regards,

Dave

KetsOne, 2021-01-25:
Same issue here. Looks like data is being sent to inverter, but no data is being sent to PVOutput.
Please help!

The log: https://pastebin.com/raw/nsvULFWQ

davevleugel, 2021-01-25:
I have an diffrent situation. 

```
(D) CRC received: 0x6 0x4C , calculated CRC: 0x6 0x4C .
(D) CRC match.
(D) New inverter found. Current # registrations: 1
(D) SendAllocateRegisterAddress address: 1
(D) Sending data to inverter(s).Sent data to inverter(s):
```

After he found my inverter it stoppes. No data to pvoutput or to mqtt broker.

When i look to your log I dont see that your inverter accepts the crc match? So you have no connection at all? 
I have an connection, but when that is established it stopes/crashes

KetsOne, 2021-01-26:
@davevleugel 
Actually i think i have the same / similar issue. The erreor above is not in my logdump, but i have encountered it a number of times in other logs.

davevleugel, 2021-01-28:
Ha ok, its to bad there is zero responds at this moment. I really cant find out what the problem is, begin last year it worked for several months.. changed the SSID after that it didnt work anymore. I tryed now 3 diffrent nodemcu's on diffrent configuration and accespoints. My last hope is to try an older version of this script.

KetsOne, 2021-01-28:
Yes, indeed.
Please let me know when you find something that works.

Ill try some forks too

Arnout 

Verstuurd vanaf mijn iPhone

davevleugel, 2021-01-28:
Ik heb het opgegeven, vandaag weer ruim 3 uur verspild maar kom altijd tot zelfde resultaat. Ik weet zeker dat hij communiceert via RS485 aangezien de omvormer het menu WIFI skipt en hij knippert traag (orange)  (volgens handleiding is dit dus OK). Hierbij heb ik ook nog een scope eraan gehangen en ik zie hem op tx als rx communiceren. Ik heb geen opties meer, volgende zal zijn om in de code te spitten, maar dat gaat in C++ net boven mijn pet.

KetsOne, 2021-02-01:
Heb nog geen forks kunnen proberen, maar ik zit ook te denken om een nieuwe compile te maken met oudere libraries (het heeft in ieder geval in het verleden goed gewerkt). En dan met name de libraries die zorgen voor connectiviteit.

Van het weekend heb ik een nieuwe firmware geflashed om te proberen of dezelfde problemen optreden...
Helaas is dit ook met de meest recente firmware het geval :(

KetsOne, 2021-02-01:
I've tried some different versions and at least is appears to collect data from the inverter. After 5 mins that data is sent to PVOutput, but immediately after that the software stops collecting data from the inverter. Consequently also the uploads stop...
So not much of an improvement.

davevleugel, 2021-02-01:
I have exactly the same behavior. i see one log entry on pvoutput and from my mqtt message like 5-10 minutes of data. Can you exclude pvoutput part, if it is working then it is fine for me, now i do it with nodered (semsportal API --> NodeRed --> PVoutput) :)

llevering, 2021-03-09:
I have had the same issue as @ferdyh & @davevleugel, I've commented out all everything after mqqtPublisher.handle() and the last yield() in the loop function in the GoodWeLogger main file and today it has been working all day. I am wondering if this is a lucky shot and bug number #16 will catch me in the end. But for those of you who still want to give it a try... :)

KetsOne, 2021-03-10:
@llevering, great that you have more success than @davevleugel and myself!
Can you please share the edited code with us?

llevering, 2021-03-11:
Still seems to work fine... In the near future (somewhere in the next week depending on time available) I will fork the repository and make the change there. So you can easily retrieve the changes.

For now this is the changed loop() function at the bottom of GoodWeLogger.ino file:
```
void loop()
{
	//update the time if set correctly
	auto timeUpdate = timeClient.update();

	//if the time is updated, we need to set it in the time lib so we can use the helper functions from timelib
	if (timeUpdate)
	{
		if (!validTimeSet)
			validTimeSet = true; //pvoutput is started after the time is set

		//sync time to time lib
		setTime(timeClient.getEpochTime());
	}

	//check wifi connection
	if (!checkConnectToWifi())
	{
		Serial.println("Wifi connection lost for too long. Restarting ESP");
		ESP.restart();
	}

	ArduinoOTA.handle();
	yield();
	goodweComms.handle();
	yield();
	mqqtPublisher.handle();
//	yield();
	//start the pvoutput publisher after the time has been set if it is configured to start
//	if (validTimeSet && pvoutputPublisher.canStart() && !pvoutputPublisher.getIsStarted())
//		pvoutputPublisher.start();

//	pvoutputPublisher.handle();
	yield(); //prevent wathcdog timeout

#ifdef REMOTE_DEBUGGING_ENABLED
	Debug.handle();
#endif
}
```

KetsOne, 2021-03-29:
@llevering 
I've tried your update and now the upload to PVOutput does not work.
Which is logical since you commented these lines out.

However, for someone who only requires MQTT this solution might be sufficient.

davevleugel, 2021-05-22:
> @llevering
> I've tried your update and now the upload to PVOutput does not work.
> Which is logical since you commented these lines out.
> 
> However, for someone who only requires MQTT this solution might be sufficient.

You already found something? It is not working for me either:(

KetsOne, 2021-08-19:
@davevleugel 
You can use the code @llevering pasted above. Sicne the lines of code for posting to pvoutput are commented out, only MQTT will work.

davevleugel, 2021-08-19:
> @davevleugel
> You can use the code @llevering pasted above. Sicne the lines of code for posting to pvoutput are commented out, only MQTT will work.

Thank you, i did not notice that part. il try it this weekend! Thanks for pointing out