jantenhove/GoodWeLogger #33: GW3000-SS

niehans, 2020-12-26:
I have the inverter without wifi module so this project is perfect for me. 
Thanks for making it possible.

I have some issues. 
First I can't figure out witch of these wires of my inverter going to the "TTL to RS485" board.
[image]

Second
How should the LEDs on the TTL board blinks? My RX is steady and my TX flashes for every 10 seconds I think

Last:
Which pins are the right ones on the NodeMCU boards?
The label on the boards says D1 and D2 but I'm not sure. 
[image]

Sorry for all the noob questions but I don't know where else to go.
Best regards
Soren

ThinkPadNL, 2020-12-28:
I think you need pin 6 & 8 (RS485**A**). On the NodeMCU just follow the labeling on the board itself, so the pins top right that have the GPIO5 & GPIO4 labels in your image.

davevleugel, 2021-01-12:
What issue do you exactly have, do you have a debuglog?

niehans, 2021-01-22:
I moved to GoodWeUSB logger instead after a million failed attempts. It works for me now. 

Best regards