jantenhove/GoodWeLogger #35: GW8K-DT inverter

dirksenrdh, 2021-01-24:
This is a very interesting project, thankt for all the information.
I own since last week a Goodwe GW8K-DT but the connections are completely different from what I see in this project.
and non of the manuals I found is completly concurrent with my hardware. On the front I have the 2 line LCD display with only ONE button. On the bottom I have this.
[image]
In the USB port is a usb-wifi dongle which opens the route to the SEMS portal and my information is loaded there.
the other two COM connectors shoud ben RS485 but why is one of them bridged whith a jumper?

I'd rather not dig into the ESP route but I'd rather use a RS485 to Ethernet converter and run the programs on my Pi 4B.
does anyone has experiance with this route?

davevleugel, 2021-01-25:
This project is based to read out the serial communication. Ill also read your intressed to use an Rpi. Maby this project (side project of this one, is maby interresting?: [https://github.com/sircuri/GoodWeUSBLogger](url)

ThinkPadNL, 2021-01-31:
Do NOT plugin a cable to the USB port without first verifying its pinout! The pinout/polary may be different, which could damage devices. See #28

dirksenrdh, 2021-02-02:
I'm a little further on this. the lower connector is a remote enable/disable inverter. if you remove the bridge the inverter switches off.
the upper connector should be RS485 with a modbus protocol.
if anyone knows the type and make of the connector I am very interested. furthermore if the upper connector is a RS485 it is interesting to know which is the A and which is the B line.

ThinkPadNL, 2021-02-02:
I don't think having the polarity wrong will damage anything. If i read [here](https://electronics.stackexchange.com/questions/397268/how-to-deal-with-signal-polarity-relative-to-usb-rs485-converters) the signal will just be inverted. So i think that if it won't work, you can just swap around the wires.

dirksenrdh, 2021-02-02:
Got a message from Goodwe BNL, they are setting up a webshop where users can order parts like the connectors. should be operational beginning of march.  (no information march what year ;-) )

mletenay, 2021-09-08:
@dirksenrdh I don't know if you have succeeded with your inverter, but maybe this reference will help.
I happen to be author of this [HomeAssistant plugin,](https://github.com/mletenay/home-assistant-goodwe-inverter) out of which was this [python library](https://github.com/marcelblijleven/goodwe) extracted.
So if you search for non-rs485 solution, the DT can be easily communicated with via WIFI and UDP protocol.

dirksenrdh, 2021-09-09:
@mletenay Yep, I bought the connector at a local dealer. then I bought a HF2211 RS485 <> ethernet converter and I wrote some Python code to retrieve the actual data about every 2 seconds. Next I push the data in a set virtual devices in my Domoticz system.
In Domotics I also read every second my P1 SmartMeter so I have an up to date view of my energy situation.
I use the wifi connection to communicate with the SEMS portal so that my PV supplier can do his health checks om my PV installation.

dirksenrdh, 2021-09-21:
Yes, In the netherlands sold by PV Care (solarservicedesk@gmail.com)
, brand is Exceedconn article 910001501 and if you need a lot of these connectors you can look here
https://exceedconn.en.alibaba.com/product/62133478391-811550881/connectors_led_strip_ip68_2_pin_male_plug_waterproof_connector.html

dirksenrdh, 2021-09-21:
or here: http://www.exceedconn.com/e_products_detail.asp?id=136
EP020-1224-00

frankhommers, 2021-09-21:
Thanks! Great. Can I just hook up an RS485 to USB connector or does it need to be terminated somehow?

dirksenrdh, 2021-09-22:
I think you have to read the RS485 specs. I have a 5 meter 2 wire twisted pair between de inverter and my HF2211 converter. no termination on either side, no problems at all.
the connection is only 9600 baud so less vulnerable to noise and reflections than the super high baudrate connections.

dirksenrdh, 2021-09-22:
eehhh, no you can't hook up an RS485 to a USB connector.
RS485 is a hardware specification for a mult-drop bus. is specifies electrical values.
USB is a complete different hardware specification and you can't combine them. Unless you use a USB formfactor connector to connect RS485 cabling. but that would be very unwise.
My inverter has an USB formfactor connector which is not USB. you only can plug in the Goodwe WIFI or Ethernet dongles.
besides this "USB" connector I have the two round plugs als shown in the picture. One of these round plugs is holding the A and B lines of a RS485 bus.
so on the other end of the RS485 cable you need to attacht a RS485 tranceiver (like the HF2211) and on top of that you need the software to handle the MODbus protocol that taks to the Goodwe inverter.

frankhommers, 2021-09-22:
Oh, I wrote connector, but I meant converter :| 
This one to be specific: https://nl.aliexpress.com/item/32428596578.html

BTW: I did not get response from PV Care yet.

dirksenrdh, 2021-09-22:
yeah, ok. I wanted to be safe.
My config is 
Goodwe -> HF2211 RS485 ethernet converter -> Pi with Python program and Domoticz
the HF2211 is located close to the inverter. the location of the pi is completely irrelevant als long as it can connect tot ethernet.
I find the drawback of you solution that RS485 and USB are both limited cable lenths. And although 485 can stretch to 1 km you have to install a dedicated cable.

frankhommers, 2021-09-22:
Yeah, I have two inverters actually, already both already having ethernet there. They are connected to the LAN modules now.
So ethernet is already there.

dirksenrdh, 2021-09-22:
so you can connect both inverters on the RS485 bus to the 485 -> USB converter.
do not forget tot change the MODbus address of at least one of the 2 units, both are default on 247 (decimal)
and ofcourse your software should be able to handle multiple modbus subscribers.

frankhommers, 2021-09-22:
I think I will just connect them to each adapter individually.

dirksenrdh, 2021-09-23:
which device in your setup is handling the MODbus communication to the inverters? is that the adapter? that would surprise me. or is it the pi with Ser2Net or is it a device furter upstream. because the MODbus software is interrogating the inverters. if you don't change the MODbus address you will experience collisions.

karolkarolski25, 2022-05-03:
@frankhommers Did you get any response? I have the same inverter with 6 pins and I think about creating my own meter. Provided WiFi adapter still emits SSID and I cannot turn it off which is anoying for me D: